dronekit/dronekit-sitl #146: Changing --home parameters in terminal doesn't work in MAC

nantha42, 2020-07-10:
I am using APM planner. When I enter dronekit-sitl copter with --home=-35.363261,149.165230,584,353 in the terminal it works fine. The copter's barometer is calibrated. But if I change the parameter to --home = 12.934158, 77.609316,584,353. The drone gets placed at some other location in the APM planner instead of the given coordinates and the barometer is not calibrating. How to solve this issue. Please respond as soon as possible. 
Thank you!

peterbarker, 2020-07-11:
On Fri, 10 Jul 2020, Nanthak wrote:

> copter's barometer is calibrated. But if I change the parameter to --home =
> 12.934158, 77.609316,584,353. The drone gets placed at some other location

Check your formatting; --home=12.934158,77.609316,584,353

nantha42, 2020-07-11:
> On Fri, 10 Jul 2020, Nanthak wrote: copter's barometer is calibrated. But if I change the parameter to --home = 12.934158, 77.609316,584,353. The drone gets placed at some other location
> Check your formatting; --home=12.934158,77.609316,584,353

Thanks it worked sir

nantha42, 2020-07-11:
Thanks it worked!

closed 2020-07-11 by nantha42