dronekit/dronekit-sitl #148: CRITICAL:autopilot:

atomattias, 2020-09-23:
Has anyone seen a problem with the simple_goto.py where it keeps showing "CRITICAL STATUS"? The copter than fails to takeoff. I ran this numerous times before with never any problems.
Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
Connecting to vehicle on: tcp:127.0.0.1:5760
CRITICAL:autopilot:APM:Copter V3.3 (d6053245)
CRITICAL:autopilot:Frame: QUAD
CRITICAL:autopilot:Calibrating barometer
CRITICAL:autopilot:Initialising APM...
CRITICAL:autopilot:barometer calibration complete
CRITICAL:autopilot:GROUND START
Basic pre-arm checks
Waiting for vehicle to initialise...
Waiting for vehicle to initialise...
Waiting for vehicle to initialise...
Waiting for vehicle to initialise...
Waiting for vehicle to initialise...
Arming motors
Waiting for arming...
Waiting for arming...
Waiting for arming...
Waiting for arming...
ERROR:autopilot:ARMING MOTORS
CRITICAL:autopilot:GROUND START

loicspace, 2020-10-05:
Same issue for me, for this script specifically and my own, which all run successfully on an older install.

JesperA75, 2020-12-18:
Which version do you use that actually works? I need to revert :-)

CarlosMuli, 2021-01-22:
I am getting the same issue has anyone found a solution?

JesperA75, 2021-01-22:
Yes!

CarlosMuli, 2021-01-22:
@JesperA75 That's a relief to hear, could you tell me how you managed to go around this issue? I thought that there might have been an issue with the version of Ubuntu that I was using but I switched to 16.04 there was seemingly no change.

JesperA75, 2021-01-23:
Simply uninstall dronekit and install version 2.9.1. sad that dronekit has
been abandoned. I guess dronecode is the new thing.....