dronekit/dronekit-sitl #149: Dronekit SITL error

Turbullance, 2020-11-12:
Companion Computer: RPi 4B
Dronekit : 2.9.2
Dronekit sitl: 3.3.0

I received this two errors:

1) pi@raspberrypi:~ $ dronekit-sitl copter
os: linux, apm: copter, release: stable
SITL already Downloaded and Extracted.
Ready to boot.
Traceback (most recent call last):
File "/home/pi/.local/bin/dronekit-sitl", line 8, in <module>
sys.exit(main())
File "/home/pi/.local/lib/python3.7/site-packages/dronekit_sitl/init.py", line 601, in main
sitl.launch(args, verbose=True)
File "/home/pi/.local/lib/python3.7/site-packages/dronekit_sitl/init.py", line 251, in launch
caps = ArdupilotCapabilities(self.path)
File "/home/pi/.local/lib/python3.7/site-packages/dronekit_sitl/init.py", line 160, in init
process = subprocess.Popen([path, '--help'], stdout=subprocess.PIPE)
File "/usr/lib/python3.7/subprocess.py", line 775, in init
restore_signals, start_new_session)
File "/usr/lib/python3.7/subprocess.py", line 1522, in _execute_child
raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/home/pi/.dronekit/sitl/copter-3.3/apm'

2) pi@raspberrypi:~ $ python hello.py
Start simulator (SITL)
Traceback (most recent call last):
File "hello.py", line 2, in <module>
import dronekit_sitl
ImportError: No module named dronekit_sitl

Turbullance, 2020-11-12:
@peterbarker

4NTH3L1OS, 2020-11-12:
Hi, You get the error because Dronekit-SITL doesn't support ARM builds.

Here's the link on the subject: https://dronekit-python.readthedocs.io/en/latest/develop/sitl_setup.html

peterbarker, 2020-11-13:
On Thu, 12 Nov 2020, 4NTH3L1OS wrote:

> Hi, You get the error because Dronekit-SITL doesn't support ARM builds.

You can make it work by passing in a binary and set of parameters to use. 
The binaries can be downloaded from firmware.ardupilot.org or compiled 
from source.

My PR here: https://github.com/dronekit/dronekit-sitl/pull/145 was 
attempting to allow automated download using ArduPilot's manifest files, 
but you might notice I haven't looked at it in some time.

Peter

4NTH3L1OS, 2020-11-13:
Yeah, sure. It can be tried. It could probably work. But 100% I'm not sure. 

I had the same problem 3-4 months ago. I ran the SITL on a normal laptop to test the codes I wrote. Obviously, if you don't have to, you'd better use normal laptop instead of Raspberry Pi.