dronekit/dronekit-sitl #150: I can't change to Vehicle Mode

zatek5, 2020-11-17:
I am using this code ;

#!/usr/bin/env python
# -*- coding: utf-8 -*-

"""
© Copyright 2015-2016, 3D Robotics.
simple_goto.py: GUIDED mode "simple goto" example (Copter Only)

Demonstrates how to arm and takeoff in Copter and how to navigate to points using Vehicle.simple_goto.

Full documentation is provided at http://python.dronekit.io/examples/simple_goto.html
"""

from __future__ import print_function
import time
from dronekit import connect, VehicleMode, LocationGlobalRelative

# Set up option parsing to get connection string
import argparse
parser = argparse.ArgumentParser(description='Commands vehicle using vehicle.simple_goto.')
parser.add_argument('--connect',
                    help="Vehicle connection target string. If not specified, SITL automatically started and used.")
args = parser.parse_args()

connection_string = args.connect
sitl = None

# Start SITL if no connection string specified
if not connection_string:
    import dronekit_sitl
    sitl = dronekit_sitl.start_default()
    connection_string = sitl.connection_string()

# Connect to the Vehicle
print('Connecting to vehicle on: %s' % connection_string)
vehicle = connect(connection_string, wait_ready=True)

while vehicle.mode.name != "POSHOLD": 
    vehicle.mode = VehicleMode("POSHOLD")
    print("in loop")
    time.sleep(0.1)
    

# Close vehicle object before exiting script
print("Close vehicle object")
vehicle.close()

# Shut down simulator if it was started.
if sitl:
    sitl.stop()

-----------------------------------
And output ; 

Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
Connecting to vehicle on: tcp:127.0.0.1:5760
CRITICAL:autopilot:APM:Copter V3.3 (d6053245)
CRITICAL:autopilot:Frame: QUAD
CRITICAL:autopilot:Calibrating barometer
CRITICAL:autopilot:Initialising APM...
CRITICAL:autopilot:barometer calibration complete
CRITICAL:autopilot:GROUND START
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
in loop
.
.
.

JesperA75, 2020-12-18:
I have the same issue - is there a known dronekit version that I can revert back to?

walchko, 2021-01-09:
same issue

Jovian-Dsouza, 2021-01-13:
I had the same issue , I solved by updating to copter-3.6 as given in #142 

Here are the steps I followed
```
cd ~/.dronekit/sitl/copter-3.3
wget https://github.com/dronekit/dronekit-sitl/files/3514761/firmware_36.zip

```

Mettikodeva-Nara, 2021-11-03:
> I had the same issue , I solved by updating to copter-3.6 as given in #142
> 
> Here are the steps I followed
> 
> ```
> cd ~/.dronekit/sitl/copter-3.3
> rm *
> wget https://github.com/dronekit/dronekit-sitl/files/3514761/firmware_36.zip
> unzip firmware_36.zip
> mv copter_36 apm
> sudo chmod -R a+rwx ~/.dronekit/sitl/copter-3.3/
> ```

how to update the copter firmware in windows?
thanks