dronekit/dronekit-sitl #151: Dronekit Plane Simulation can not be started

oskar07, 2021-03-19:
Hello everyone, I was trying to test dronekit examples on ubuntu 16.04 I runned the example codes succesfully for the copter vehicle by using dronekit sitl. When I moved into trying same example codes on plane vehicle I couldnt succed to run sitl.  I am running on first terminal:
dronekit-sitl plane
on the second terminal:
mavproxy.py --master tcp:127.0.0.1:5760 --out 127.0.0:14550 --out 127.0.0.1:14551
and I am getting :
Connect tcp:127.0.0.1:5760 source_system=255
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
 MAV> APM: Calibrating barometer
online system 1
INITIALISING> Mode INITIALISING
APM: Initialising APM...
PANIC: AP_Baro::read unsuccessful for more than 500ms in AP_Baro::calibrate [2]

no link
link 1 down
no link
no link
no link
no link
no link
no link
no link
no link
no link
no link
no link

I would really appriciate any help regarding to this issue.

pappuyadav, 2021-03-20:
@oskar07 Please check out the tutorial at the link below.
https://www.youtube.com/watch?v=HD51_zvBn6U

oskar07, 2021-03-25:
Thank you for your reply @pappuyadav . The video you shared is about drones, and I can succesfully made it work. The problem that I am getting is for planes.

RotorBoy2020, 2021-03-28:
@oskar07 Any update? 

I am also trying to get the plane simulation working. The drone sims work as expected but whenever I try and connect to the plane sitl I get: 
`WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds`

ruchikmishra, 2021-07-09:
Hello,
I am also facing the same issue with planes.  Copter runs fine for me.  Did you find a solution for planes? @oskar07 @RotorBoy2020.

Pascal91, 2022-04-16:
Hello,

I have the same "blocking" issue:
"`WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds`"

Code on simulator (drone) side:
```
import dronekit_sitl
print("Starting simulator...")
simulator = dronekit_sitl.start_default()
```

Code on station side:
```
import dronekit
print("Connecting to the drone...")
drone = dronekit.connect(input("Port ? "), wait_ready=True, baud=57600)
```

_If I remember well, it was working correctly with Python 3.9, but now fails with 3.10._

Thanks,
Best regards,
Pascal.

ocetin, 2022-06-04:
You are right, there is a bug for plane setup in dronekit-sitl. If I set "drone" I can connect dronekit-sitl via ardupilot mission planner successfully, but if I choose "plane" it is not working.

I am trying to solve the problem during last 2 days. Please inform me if anyone manage to solve it, thanks...

kalender26, 2022-11-28:
> Haklısınız, dronekit-sitl'de uçak kurulumuyla ilgili bir bug var. "Drone" ayarlarsam dronekit-sitl'i ardupilot görev planlayıcısı aracılığıyla başarıyla bağlayabilirim, ancak "plane" seçersem çalışmıyor.
> 
> Son 2 gündür sorunu çözmeye çalışıyorum. Çözebilen varsa lütfen bilgilendirsin teşekkürler...

Hello. Did you solve the plane problem?

metehanacikgoz, 2022-12-16:
> 
Hello.  Did you find a solution for the problem?

raed23c, 2023-02-15:
I am having the same problem. Did anyone find a solution?

pappuyadav, 2023-02-15:
Hello All,
For issue like "WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds", I would recommend to manually set the "Guided" mode from Mission Planner. Please see if that helps. 
Thanks.