dronekit/dronekit-sitl #153: Cannot Change Flight Mode on DroneKit-SITL from Command Line

ashutoshraiofficial, 2021-07-02:
I have been building a simple navigation (“go from Point A to Point B”) tool using Python 3 and DroneKit (with DroneKit-SITL for testing for now).

I set up DroneKit-SITL and MAVProxy on my Mac computer, and while I can get everything to configure, I cannot change the copter’s flight mode from “STABILIZE” to “GUIDED” (or any other mode) using the MAVProxy command line interface. Funnily enough, when I use DroneKit-SITL with Python to input commands (instead of the command line) such as changing the copter’s flight mode and navigating, it works.

I was wondering what might be causing the problem in changing the DroneKit-SITL copter’s mode from the command line? It seems that other people have had the same problem (https://github.com/dronekit/dronekit-sitl/issues/142 ) but I couldn’t find a solution that works. The suggestion to add firmware and rename “copter_36” to “apm” from the linked forum did not work on my Mac either.

I hope we can collaborate and find a solution — thanks!

viaSeunghyun, 2023-10-10:
Interestingly, or unfortunately, the simplest workaround for most mode change issues is to use python2.

It appears to have been recently ported to support python3 via 2to3, but there are still issues with python3 and it is not officially supported.

i-vis, 2024-08-09:
There is a fix in PR #145, although you might need to modify some lines in that PR. After changing .venv/lib/python3.12/site-packages/dronekit_sitl/__init__.py it should download latest SITL binary and issues like this get fixed. It worked for me and downloaded new files to ~/.dronekit/sitl/copter-stable/