dronekit/dronekit-sitl #155: Dronekit Python not connecting with SITL Plane (Copter system works)

KrishnanS2006, 2022-01-21:
Hello,

I am trying to start a Dronekit SITL simulation using the API, and connect to it with Dronekit-Python. Here is the first part of my code:
```py
from dronekit import connect, VehicleMode
from dronekit_sitl import SITL, start_default

sitl = SITL()
sitl.download("plane", "3.3.0", verbose=True)
sitl.launch([], await_ready=True, restart=True, verbose=True)
connection_string = sitl.connection_string()

vehicle = connect(connection_string)
```

This code works perfectly when using `sitl.download("copter", "3.3", verbose=True)` as shown in the example. However, the code that starts a plane simulation does not work like so, and there is a connection timeout. 

Here is the full traceback:
```
SITL already Downloaded and Extracted.
Ready to boot.
Execute: /home/krishnan/.dronekit/sitl/plane-3.3.0/apm --home=-35.363261,149.165230,584,353 --model=plane -I 0
SITL-0> Starting sketch 'ArduPlane'
SITL-0.stderr> bind port 5760 for 0
SITL-0> Starting SITL input
SITL-0.stderr> Serial port 0 on TCP port 5760
SITL-0> Waiting for connection ....
SITL-0.stderr> bind port 5762 for 2
SITL-0.stderr> Serial port 2 on TCP port 5762
SITL-0.stderr> bind port 5763 for 3
SITL-0.stderr> Serial port 3 on TCP port 5763
WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds
ERROR:dronekit.mavlink:Exception in MAVLink input loop
Traceback (most recent call last):
  File "/venv/lib/python3.9/site-packages/dronekit/mavlink.py", line 211, in mavlink_thread_in
    fn(self)
  File "/venv/lib/python3.9/site-packages/dronekit/__init__.py", line 1370, in listener
    raise APIException('No heartbeat in %s seconds, aborting.' %
dronekit.APIException: No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "test.py", line 9, in <module>
    vehicle = connect(connection_string)
  File "/venv/lib/python3.9/site-packages/dronekit/__init__.py", line 3166, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/venv/lib/python3.9/site-packages/dronekit/__init__.py", line 2275, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.

Process finished with exit code 1
```

I've already tried:
 - Increasing the `timeout` and `heartbeat_timeout` in the `connect()` function
 - Manually defining the connection string (`127.0.0.1:5760`)

Is there any connection parameter, setting, or option (plane-specific) that I am missing? Any help is appreciated, Thanks a lot!

Pexeus, 2022-02-06:
I have  the exact same issue. Any progress on this?

KrishnanS2006, 2022-02-11:
Update: I have narrowed down the problem. It occurs when specifying the plane's *model*. Here is some example code:
```py
from dronekit_sitl import SITL

sitl = SITL()
sitl.download('plane', '3.3.0', verbose=True)
sitl_args = ['--model', 'quad', ]
sitl.launch(sitl_args, await_ready=True, restart=True)
```
And the output (when running [this](https://dronekit-python.readthedocs.io/en/latest/guide/quick_start.html)):
```
SITL already Downloaded and Extracted.
Ready to boot.
Execute: /home/krishnan/.dronekit/sitl/plane-3.3.0/apm --model quad --home=-35.363261,149.165230,584,353 -I 0
SITL-0> Started model quad at -35.363261,149.165230,584,353 at speed 1.0
SITL-0.stderr> bind port 5760 for 0
SITL-0> Starting sketch 'ArduPlane'
SITL-0.stderr> Serial port 0 on TCP port 5760
SITL-0> Starting SITL input
SITL-0> Waiting for connection ....
Connecting to vehicle on: tcp:127.0.0.1:5760
SITL-0.stderr> bind port 5762 for 2
SITL-0.stderr> Serial port 2 on TCP port 5762
CRITICAL:autopilot:Calibrating barometer
SITL-0.stderr> bind port 5763 for 3
SITL-0.stderr> Serial port 3 on TCP port 5763
CRITICAL:autopilot:Initialising APM...
CRITICAL:autopilot:Initialising APM...
CRITICAL:autopilot:Ready to FLY.
CRITICAL:autopilot:init home
CRITICAL:autopilot:gps alt: 58400
CRITICAL:autopilot:ArduPlane V3.3.0 (d6053245)
CRITICAL:autopilot:command received:
CRITICAL:autopilot:command received:
CRITICAL:autopilot:command received:
CRITICAL:autopilot:command received:
CRITICAL:autopilot:command received:
CRITICAL:autopilot:command received:
Get some vehicle attribute values:
 GPS: GPSInfo:fix=3,num_sat=10
 Battery: Battery:voltage=12.587,current=0.0,level=100
 Last Heartbeat: 0.2501799449964892
 Is Armable?: False
 System status: STANDBY
 Mode: MANUAL
Completed
```

This works, but only because I have specified the `model` to be `quad`. When replacing this with `plane`, I get the issue described above. 

I assume this is a bug with the `plane` model specifically, as when I run `dronekit-sitl plane-3.3.0` (which assumes a `plane` model), I still get the heartbeat error when connecting. When I instead run `dronekit-sitl plane-3.3.0 --model quad`, by specifying the model to be `quad` it works perfectly.

ocetin, 2022-06-04:
> Update: I have narrowed down the problem. It occurs when specifying the plane's _model_. Here is some example code:
> 
> ```python
> from dronekit_sitl import SITL
> 
> sitl = SITL()
> sitl.download('plane', '3.3.0', verbose=True)
> sitl_args = ['--model', 'quad', ]
> sitl.launch(sitl_args, await_ready=True, restart=True)
> ```
> 
> And the output (when running [this](https://dronekit-python.readthedocs.io/en/latest/guide/quick_start.html)):
> 
> ```
> SITL already Downloaded and Extracted.
> Ready to boot.
> Execute: /home/krishnan/.dronekit/sitl/plane-3.3.0/apm --model quad --home=-35.363261,149.165230,584,353 -I 0
> SITL-0> Started model quad at -35.363261,149.165230,584,353 at speed 1.0
> SITL-0.stderr> bind port 5760 for 0
> SITL-0> Starting sketch 'ArduPlane'
> SITL-0.stderr> Serial port 0 on TCP port 5760
> SITL-0> Starting SITL input
> SITL-0> Waiting for connection ....
> Connecting to vehicle on: tcp:127.0.0.1:5760
> SITL-0.stderr> bind port 5762 for 2
> SITL-0.stderr> Serial port 2 on TCP port 5762
> CRITICAL:autopilot:Calibrating barometer
> SITL-0.stderr> bind port 5763 for 3
> SITL-0.stderr> Serial port 3 on TCP port 5763
> CRITICAL:autopilot:Initialising APM...
> CRITICAL:autopilot:Initialising APM...
> CRITICAL:autopilot:Ready to FLY.
> CRITICAL:autopilot:init home
> CRITICAL:autopilot:gps alt: 58400
> CRITICAL:autopilot:ArduPlane V3.3.0 (d6053245)
> CRITICAL:autopilot:command received:
> CRITICAL:autopilot:command received:
> CRITICAL:autopilot:command received:
> CRITICAL:autopilot:command received:
> CRITICAL:autopilot:command received:
> CRITICAL:autopilot:command received:
> Get some vehicle attribute values:
>  GPS: GPSInfo:fix=3,num_sat=10
>  Battery: Battery:voltage=12.587,current=0.0,level=100
>  Last Heartbeat: 0.2501799449964892
>  Is Armable?: False
>  System status: STANDBY
>  Mode: MANUAL
> Completed
> ```
> 
> This works, but only because I have specified the `model` to be `quad`. When replacing this with `plane`, I get the issue described above.
> 
> I assume this is a bug with the `plane` model specifically, as when I run `dronekit-sitl plane-3.3.0` (which assumes a `plane` model), I still get the heartbeat error when connecting. When I instead run `dronekit-sitl plane-3.3.0 --model quad`, by specifying the model to be `quad` it works perfectly.

You are right, there is a bug for plane setup in dronekit-sitl. If I set "drone" I can connect dronekit-sitl via ardupilot mission planner successfully, but if I choose "plane" it is not working.

Besides if I attached --model plane, Mission Planner can connect SITL but most of the parameters are wrong for plane ....

I am trying to solve the problem during last 2 days. Please inform me if anyone manage to solve it, thanks...

NaheelSalman, 2023-02-08:
for my case running command:
dronekit-sitl --reset 
then normal:
dronekit-sitl copter-3.3
Works

nxxvaa, 2023-04-26:
I have the same issue with the latest update to ardupilot repo any updates on the plane issue?

Varun-UTS, 2024-08-01:
Hi has anyone solved the problem, I was facing the same issue.

hamishwillee, 2024-08-01:
Probably not. No one is maintaining this.