dronekit/dronekit-sitl #156: How can I run Dronekit-SITL on systems with "ARM" processors?

developeralioz, 2022-04-10:
I am using a MacBook with Apple M1 processor. I installed Ubuntu 20.04 ARM version on the virtual machine. I did all the installations correctly. But Dronekit-SITL is not working. I am sharing the error I got below.

`parallels@ubuntu-linux-20-04-desktop:~$ dronekit-sitl copter-3.3 --home=35.9835973,-95.8742309,0,180
os: linux, apm: copter, release: 3.3
Downloading SITL from http://dronekit-assets.s3.amazonaws.com/sitl/copter/sitl-linux-copter-3.3.tar.gz
Download Complete.
Payload Extracted.
Ready to boot.
Traceback (most recent call last):
  File "/usr/local/bin/dronekit-sitl", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/dronekit_sitl/__init__.py", line 601, in main
    sitl.launch(args, verbose=True)
  File "/usr/local/lib/python3.8/dist-packages/dronekit_sitl/__init__.py", line 251, in launch
    caps = ArdupilotCapabilities(self.path)
  File "/usr/local/lib/python3.8/dist-packages/dronekit_sitl/__init__.py", line 160, in __init__
    process = subprocess.Popen([path, '--help'], stdout=subprocess.PIPE)
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/home/parallels/.dronekit/sitl/copter-3.3/apm'`

[image]

How can I run Dronekit-SITL on ARM systems?
I read something called compiling from source in other issues. Can you help with this?

hamishwillee, 2022-04-10:
Yes, you have to build the source since prebuilt binaries for ARM are not provided. There is some information on this https://dronekit-python.readthedocs.io/en/latest/develop/sitl_setup.html

There is a note there that explains how, after building, you can then run the binaries from SITL

> You can also use dronekit-sitl to start a SITL executable that you have built locally from source. To do this, put the file path of the target executable in the SITL_BINARY environment variable, or as the first argument when calling the tool.

peterbarker, 2022-10-11:
On Sun, 10 Apr 2022, Hamish Willee wrote:

> Yes, you have to build the source since prebuilt binaries for ARM are not provided. There is some information on this https://dronekit-python.readthedocs.io/en/latest/develop/sitl_setup.html

Actually... we have for about three years :-)

https://firmware.ardupilot.org/Copter/beta/SITL_arm_linux_gnueabihf/

> There is a note there that explains how, after building, you can then run the binaries from SITL
>
>       You can also use dronekit-sitl to start a SITL executable that you have built locally from [...]

There's also a very experimental branch that can retrieve the binaries 
from the firmware server: 
https://github.com/peterbarker/dronekit-sitl/commits/wip/firmware-server

hamishwillee, 2022-10-11:
Thanks for the clarification @peterbarker - I'll remember this for next time. You could update the docs.

closed 2022-10-11 by hamishwillee